nraynaud/webgcode #4: Getting Data from G Code

alexsanjoseph, 2016-03-07:
Wonderful web application! Is there a way to get the data used for plotting the graph from your app? Thanks

nraynaud, 2016-03-07:
Thank you. I have troubles understanding your question, could re-formulate it or give me some context please? There is a "Gcode" button in the top left corner of the app, does that answer your question ?

alexsanjoseph, 2016-03-07:
Hi Raynaud, thanks for the quick reply.

I am looking for a G code data simulator to map G code and actual machine raw data. I see that your app is internally converting it G-Code into the tool path, but how do I get the data out of it? Can I get the data into something like the X, Y, Z coordinates and the Speeds/Feeds/Tool information etc?

nraynaud, 2016-03-07:
well, I have not developed that. What kind of file format do you envision?

alexsanjoseph, 2016-03-07:
Please take a look at this. 
https://www.npmjs.com/package/gcode-toolpath.

`[ { motion: 'G0',
    v1: { x: 0, y: 0, z: 0 },
    v2: { x: 0, y: 0, z: 6.35 } },
  { motion: 'G0',
    v1: { x: 0, y: 0, z: 6.35 },
    v2: { x: -12.7, y: 0, z: 0 } },
  { motion: 'G0',
    v1: { x: -12.7, y: 0, z: 0 },
    v2: { x: 0, y: 0, z: 2.54 } },
  { motion: 'G1',
    v1: { x: 0, y: 0, z: 2.54 },
    v2: { x: 0, y: 0, z: 0 } },
  { motion: 'G2',
    v1: { x: 0, y: 0, z: 0 },
    v2: { x: 0, y: 12.7, z: 0 },
    v0: { x: 12.7, y: 0, z: 0 } },
  { motion: 'G2',
    v1: { x: 0, y: 12.7, z: 0 },
    v2: { x: 12.7, y: 0, z: 0 },
    v0: { x: 0, y: 0, z: 0 } },
  { motion: 'G2',
    v1: { x: 12.7, y: 0, z: 0 },
    v2: { x: 0, y: -12.7, z: 0 },
    v0: { x: 0, y: 0, z: 0 } },
  { motion: 'G2',
    v1: { x: 0, y: -12.7, z: 0 },
    v2: { x: -12.7, y: 0, z: 0 },
    v0: { x: 0, y: 0, z: 0 } },
  { motion: 'G1',
    v1: { x: -12.7, y: 0, z: 0 },
    v2: { x: 0, y: 0, z: 2.54 } },
  { motion: 'G0',
    v1: { x: 0, y: 0, z: 2.54 },
    v2: { x: 0, y: 0, z: 6.35 } } ]
`

But this is fairly basic, unlike your app which can handle complex paths. And it internally has all the information required already. I was wondering if there was an easy way to expose it. Or would you would know another project that does this?

nraynaud, 2016-03-07:
ok, you want the parsed G-code ? 
you can call evaluate() in here: https://github.com/nraynaud/webgcode/blob/gh-pages/webapp/cnc/gcode/parser.js

'text' is the g-code text.

alexsanjoseph, 2016-03-07:
Thanks. I think that might answer the question, but I need to understand basics of javascript to figure out how to run that. I will get back to you once that is figured out!

closed 2016-03-07 by nraynaud