nraynaud/webgcode #5: WebGcode simulator

pauljay, 2016-06-27:
Hi,
I have tried out webgcode simulator. But I can't paste my gcode onto the left window since there is no left window to put the gcode.

How can I make it show the left window to put my g code ?

I am making my own CNC board. My firmware is in between GRBL and Smoothie also something like a mashup of them. Though I have a g code viewer on Windows OS, I need to make a some G code community site where popular milling operations G code are exchanged like Thingiverse.

Regards,

Paul J

hanbitmicro(at)gmail.com

nraynaud, 2016-06-29:
Hi Paul,
I'm sorry an SVG API has been removed from Chrome, and you are right, it broke the left hand side panel. I'll try to fix it soon. In the mean time, you can try with firefox.

Thanks for trying my software.

Nicolas.

pauljay, 2016-06-30:
Thank you for your reply.

nraynaud, 2016-07-03:
Hello Paul, that seems to be fixed, albeit the thing is now quite slow.

closed 2016-07-03 by nraynaud

pauljay, 2016-07-03:
Yes, it works now on Chrome albeit it's some slow to load.
However, it works very flawlessly on FireFox.
Do you use WebGL on FireFox ?

nraynaud, 2016-07-03:
I use webgl on both of them. It's the SVG system that as now become slow on Chrome, since they removed a native API, it is simulated in javascript.