nraynaud/webgcode #6: VisualCam manual ?

pauljay, 2016-07-03:
Visualcam on Firefox works nice. The UI is pretty intuitive. I've seen the gcode generated by VisualCam on webgcode viewer correctly.
 Is there any online manual of VisualCam ?

nraynaud, 2016-07-03:
No there is no manual, I just do that on my own. Maybe it's time to make this project a bit more serious.

luizabbadia, 2017-04-30:
Thanks for your code.I can even see the drill gcode in your app, and I could not see it in another app I´ve tested. It helped me a lot! Thanks!!!!

nraynaud, 2017-04-30:
Thank for your kind comment. it means a lot.

luizabbadia, 2017-06-16:
Hi! It´s me again...may be you didn´t notice but I merged 3 gcodes ( etch,holes and outline( and I saw all 3 together!!! Amazing!!! It is a good diagnostic to check for overlapping codes!!! Congrats ...again!!!