nraynaud/webgcode #7: Fork for Vanity (:

openhardwarecoza, 2016-07-18:
Hey.

You have quite an interesting project, and VisualCam in particular is interesting.

Would you be willing to give me an open source fork to work from? 
I am in particular interested in extending your dragknife support, and more 3D milling. 

We also use a threejs+clipperjs backend on Laserweb

Laserweb has quite a bit of traffic - which could help bring some fame.  (We even have a sponsors WIKI page that has managed to nett some of the contributors machines, boards and paypal donations - i'd be happy to put your details on there if we use some of your code. )

nraynaud, 2016-07-18:
yeah, sure! what licence do you want?
Would MIT do it for you?
I guess it's time for me to *really* open source this stuff.

openhardwarecoza, 2016-07-18:
(: I honestly an stuck in the same boat you are - licences are sooo tough to choose. 

I have LaserWeb as GNU Affero General Public License - but there are sections of the code from contributors, forks and libraries which means its not fully GPL at all. 
My argument is I'm happy to maintain a more restrictive licence on a component inside the project, if thats the original authors wish.  Ultimately I believe credit, honor and trust come above all. 

Thats why I like sticking credits and "Author: XXX" into headers and comments so often  (:

At the end of the day - the more people see your code right (;

nraynaud, 2016-07-18:
Here we go, double license, MIT + Affero GPL.

closed 2016-07-18 by nraynaud

openhardwarecoza, 2016-07-18:
Thanks (: ! (;  

Your functions: 
[image] 

much better than what we had as work in progress ( https://github.com/openhardwarecoza/LaserWeb2/commits/DragKnife )

openhardwarecoza, 2016-07-18:
(: Do you have a Google+ account so I can tag you properly?

nraynaud, 2016-07-18:
I do have a google+ account, but I have no idea of the URL. maybe that: https://plus.google.com/u/0/103823220772217784777/ ?

nraynaud, 2016-07-18:
After thinking for about it for a bit, I think the picture has a problem, I might have made a bug.

openhardwarecoza, 2016-07-18:
Or my porting of your code may have a bug.  Transplanted just the dragknife
code into my existing code

openhardwarecoza, 2016-07-18:
https://github.com/openhardwarecoza/LaserWeb3/commit/48cac8f5d65f41013f497052516153f70230da3b