nraynaud/webgcode #10: Circular interpolation arcs end early

slindberg, 2016-12-30:
Thanks for the awesome tool, I'm using it to proof my G-code before trying it on a real CNC. I don't know if I'm missing something, but it appears that circular interpolation endpoints aren't completed properly when followed by a G0 or a G1. I think it's easiest to just show you [an example](https://gist.github.com/slindberg/c95765b1d42619cdb77d97610dc7d8c3). Pasting that code into the simulator, you can see that arcs in the XY plane terminate early, and the G0 to the clearance plane (Z0.25) is diagonal instead of vertical. This happens on the bottom of the eyes as well, when a G2 is followed by a G1.

nraynaud, 2016-12-30:
Thanks for the report with the nice repro, that looks buggy to me too.

nraynaud, 2016-12-30:
Can you check that I fixed it please?

slindberg, 2016-12-31:
Looks like that did it! Thanks again for the great tool.

commit 165dd8ce (2016-12-30): fix #10

it looks like "<=" with an index loop doesn't work very well when the cap is not an integer.
closed 2016-12-31 by slindberg