nraynaud/webgcode #11: Very useful tool!

TDeagan, 2017-01-09:
M. Raynaud,

I have recently been using your online G-code simulator and would like to request your permission to use screenshots from it in a book I am writing for MakerMedia (Make:).  The book is about using digital fabrication tools like 3D Printers, CNC, laser engravers, vinyl cutters, etc. for leatherwork.  I would like to reference your site in the book as a tool for users.  The book is coming out in Sept. of 2017.

I have previously published 'Make:Fire, The Art and Science of Working with Propane' for MakerMedia (https://www.amazon.com/Make-Fire-Science-Working-Propane/dp/1680450875) and am a regular contributor to Make: magazine (http://makezine.com/author/tim-deagan/).

Please let me know if you have any interests or concerns.  I will only use screenshots from the site with your permission.

Yours,
--Tim

nraynaud, 2017-01-09:
Hi Tim,

You are most welcome to do so. 
I think I will impress quite some people at the bar :)

I am curious, what is your use-case? a quick preview of the toolpath before actually running it?

Nicolas.

TDeagan, 2017-01-09:
Wonderful!  

I was having a terrible problem with the laser engraver software converting DXFs into G-code.  The resulting code was jumping all over the place.  When I plugged it in your simulator, it made the spaghetti visible.  Some of this had to do with the order of operation that the items were added to the DXF.  Some of it was cleaned up with a DXF optimization function.  I couldn't get my 3D printer G-code simulators to run the laser engraver g-code, so your tool was a big help.  

I'd like to show my readers the impact of different DXF construction and optimization activities on their resultant G-code, so using screenshots from your tool will be fantastic!

Thanks again, I'll keep you informed and send you review chapters when I'm able if you like.

--Tim

closed 2017-01-09 by TDeagan

nraynaud, 2017-01-09:
oh yeah, polyline order!
I have to say that my toolpath generator has the same problem, the SVG paths have to be in the correct order to not generate a stupid path. Maybe I should write an optimizer for that, it's actually quite easy to sort out.

I would be glad to read the thing before everyone else, but I don't feel entitled to it just because you grabbed 2 screenshot from my work :)

Nicolas.