nraynaud/webgcode #13: G02 with only Y parameter locks up webgcode

dafzor, 2017-03-24:
When writing the code `G02 Y20` simulating it will lock webgcode tool in a permanent simulation cycle.

nraynaud, 2017-03-25:
Thanks for your report, 
I confirm I can reproduce this behavior, let me have a closer look.

nraynaud, 2017-03-25:
Upon further inspection, I don't think that code is legal, because there is no way to compute the arc path without either a center or a radius, I'll just detect and report the error.

nraynaud, 2017-03-25:
Ok, I detected and reported the error. 
If you think I misunderstood either your bug report or how the G2 command is supposed to work, please re-open with your comments.

closed 2017-03-25 by nraynaud

dafzor, 2017-03-25:
The issue is not that the code is legal. It's that inputing the code in that manner soft locks the webgcode app into a simulation loop instead of simply trowing an error, thus forcing a reload to continue using it.

I'm currently learning gcode and using your web app to validate my results, thus the high probability of bad code inputs.

nraynaud, 2017-03-25:
ok, it's fixed for this instance.
But expect a lot more issues of the same type, I did almost no verification of the input.

dafzor, 2017-03-25:
I'll report them as I find them if you don't mind.

nraynaud, 2017-03-25:
Yes, thank you, that will increase the general quality.