nraynaud/webgcode #15: webgcode app doesn't seem to recognise valid G41, G42 gcode

dafzor, 2017-03-25:
When using the instruction `G42 D1` the webgcode app seems to not recognize the command. This however should be valid gcode (at least it seems to function as expected in the CNC machine I'm learning with).

Note however that everything still seems to function well so it's just a "minor" bug.

nraynaud, 2017-03-25:
Yes, I didn't implement it. 
Tool radius compensation is a tricky subject, some machines will move on the side as soon as you set it and other will start moving sideways on the first move after a tool radius definition. Since I had no real target in mind, I didn't bother implementing it.

dafzor, 2017-03-25:
Would it be possible to change the error message to something like "G42, Not implemented, skipping" or similar error message to make it clearer that it's not a user error?

nraynaud, 2017-03-25:
ok, done.

closed 2017-03-25 by nraynaud