nraynaud/webgcode #19: Arc in wrong direction

dastultz, 2017-12-13:
The following g-code has a 90 degree arc from Z to X. It looks like the arcs are going in the wrong direction:

%
(1001)
(T1  D=0.375 CR=0 - ZMIN=-0.075 - flat end mill)
G90 G94
G17
G20

(365 face)
M9
T1 M6
S2000 M3
G54
M9
G0 X0.5228 Y-0.1828
Z0.2
Z0.05
G1 Z-0.0375 F8
G18 G3 X0.4853 Z-0.075 I-0.0375
G1 X0.3125
X0.1625 F12
G17 G2 Y0.0852 J0.134
G1 X0.3125
G2 Y-0.1828 J-0.134
G1 X-0.1625
X-0.3125
G2 Y0.0852 J0.134
G1 X-0.1625
G18 G2 X-0.125 Z-0.0375 K0.0375 F8
G0 Z0.2
G17
M9
M30
%

nraynaud, 2017-12-14:
Hi 
thanks for the report.

I agree that something strange is afoot here. I'll investigate in the upcoming days.

nraynaud, 2017-12-14:
I think I have fixed it. can you confirm please.

dastultz, 2017-12-14:
Looks good!

closed 2017-12-14 by dastultz