nraynaud/webgcode #26: Feature to upload files?

andrewerrington, 2019-07-15:
Great software! Unless I'm missing something I can only view my output if I paste my code into the text box on the website. Would it be possible to have an 'Upload' or 'Open File' button so that I can select my file on my PC and have it uploaded and rendered?

Thanks!

nraynaud, 2019-07-15:
Hi, this project is a bit dead, but know that you can drop a gcode file on the page to load it. 

I guess I was always testing by drag'n'drop and forgot to put an actual "open" button at the time.

andrewerrington, 2019-07-15:
That is a satisfactory alternative. Thanks! I didn't know it was a feature. Could you mention it on the web page text? Where it says "Paste your G-Code..." add something like "or drag and drop your file onto this page".

Sorry to hear the project is a bit dead. It is perfect for my use. The other good one is ncviewer, but I have an outstanding bug report there that doesn't seem to be moving. Thanks for your quick response. I guess you don't need to do anything. :)

closed 2019-07-15 by andrewerrington