nraynaud/webgcode #27: Does not understand G53

kaovilai, 2019-09-25:
Here's a sample snippet

O00060
N1 (G71 TYPE I O.D. ROUGHING)
G53 G00 X0. Z0. T0
T101 (O.D. TOOL . x .031 TNR)
G50 S3200
G97 S500 M03
G54 G00 X3.2 Z0.1 M08
G96 S420
Z0.005
G01 X-0.063 F.008
G00 X3.2 Z0.1
G71 P10 Q20 U0.01 W0.005 D0.12 F0.01
N10 G00 G42 X1.4
G01 Z0. F0.006
X1.5
G03 X1.75 Z-0.125 R0.125
G01 Z-2.5
G02 X2.25 Z-2.75 R0.25
G01 X2.5
G03 X3.0 Z-3.0 R0.25
G01 Z-4.125 F.004
N20 G40 X3.2 F.02
G97 S500 M09
G53 G00 X0. Z0. T0
M01

nraynaud, 2019-09-25:
you are right, G53 should be here: https://github.com/nraynaud/webgcode/blob/gh-pages/webapp/cnc/gcode/parser.js#L55

The issue is that I have no concept of machine in the simulator, so machine origin is a sketchy thing.

SlawomirUrbaniak, 2020-05-28:
Hello !
My congratulations and admiration - Super work this CNC Simulator
Nowadays, pandemics are a great tool for students
For distance learning, it's great because it doesn't require installation, which often causes problems for students of mechanical technical secondary school :-)

SlawomirUrbaniak, 2020-05-29:
Hi, 
I have a question :
it would be a lot of work to add G18 as a CNC lathe
In this case, the value of the X axis would be divided into 2
And the 2D simulation window layout would change from G17 to G18 ...
I know this requires work but it would help students to better understand CNC lathe programming ...
Technical teacher at a high school in Poland :-)