nraynaud/webgcode #31: Feature Request: Support for Marlin G5 Cubic Bezier Spline

herbertp, 2019-11-06:
It would be great to have support for the G5 command provided by the Marlin Firmware (http://marlinfw.org/docs/gcode/G005.html) 

Thanks in advance,
Herbert

nraynaud, 2019-11-14:
Hi, this project is a bit on pause at the moment. But I don't want to say no outright. 
I propose you tackle the the issue, and I will support you until it's merged. There is some Bezier code here: https://github.com/nraynaud/webgcode/blob/gh-pages/webapp/cnc/maths/bezier.js  .

But it's not a rasterization process, so it will probably start with the google query "bresenham bezier" here is a webpage http://members.chello.at/~easyfilter/bresenham.html

cheers, 
Nico

herbertp, 2019-11-14:
Hello Nico,

> Hi, this project is a bit on pause at the moment. 

Sorry to hear, because it seems quite useful to me.
One of the best online viewer I found so far.

> But I don't want to say no outright. 

Great!

> I propose you tackle the the issue, and I will 
> support you until it's merged. 

Fair enough, my main problem so far is finding the
code responsible for drawing and to recreate a local
version of your codebase.

That should be something you can help with easily :)

> There is some Bezier code here: 
> https://github.com/nraynaud/webgcode/blob/gh-pages/webapp/cnc/maths/bezier.js  .

> But it's not a rasterization process, so it will 
> probably start with the google query "bresenham 
> bezier" here is a webpage 
> http://members.chello.at/~easyfilter/bresenham.html

I propose to re-use the code present in Marlin which
will not only handle the bezier curve but also the
segmentation into lines, which is probably the way
to go to get a good result for the preview ... please
correct me if I'm wrong here :)

Thanks in advance,
Herbert

nraynaud, 2019-11-14:
You probably will need to add a case to this `if` here:
https://github.com/nraynaud/webgcode/blob/f3ad16371fc8c709f328a16dc728a256438b4d1b/webapp/cnc/gcode/gcodeSimulation.js#L9

and add a type to COMPONENT_TYPES. Actually you will need to do the job twice, once in Bresenham to compute the stepper motor steps (and that give you the timing), and once in arc segments to compute the display.

The display part is a map "code source line" -> "polyline" called `lineSegmentMap ` (you can search across the code base) so that when you click on a line in the code, it higlights it in the 3D view.