nraynaud/webgcode #39: Offline version of g-code simulator

kebjs, 2022-09-29:
> There is a g-code simulator here: http://nraynaud.github.io/webgcode/ 

Is there a way to have the g-code simulator run from disk (offline)?

Simply downloading it and starting it does not work:
- it looses all the interactions in the "black" areas (canvas, svg, ...)
- it stores the pasted g-code
- but cannot paste anything else

Any hints?

Downloaded (offline)
[image]

Online
[image]

nraynaud, 2022-09-29:
yes, Workers are absolutely forbidden from loading files from the local disk:
https://stackoverflow.com/questions/21408510/chrome-cant-load-web-worker

Basically, to avoid mischievous javascript, all files loaded from `file://` are considered as having a different origin, and a lot of things are forbidden accros origins. You best bet is to use a small HTTP server locally. Very often browsers will allow you to use http:// for restricted APIs, if the IP address resolves to localhost.

kebjs, 2022-09-29:
Thanks for your quick response; a very short test gave the same results (in a node.js environment).
I'll let you know if I manage to pull this off.

closed 2022-09-29 by kebjs